frangoteam/FUXA #150: RTU write error

danlesbm, 2021-05-18:
Hi.
I tested function RTU, but it's not error. Photo attatch
[image]

unocelli, 2021-05-20:
Hi,
to be honest I have never tested RTU but I can try to help you...with which other client it works correctly?

danlesbm, 2021-05-21:
> to be honest I have never tested RTU but I can try to help you...with which other client it works correctly?

Thanks. RTU works fine in read mode, but write mode it's wrong as picture. It can''t write to the device's register.
with this function, TCP can work well

unocelli, 2021-05-29:
I'm a bit busy at the moment, I'll take care of it as soon as possible.

alpaslanske, 2021-07-02:
I go inside this problem. What i found is below.

I used a servo drive with Rs485 connection at 19200bps, 8N1 standard connection.

first i couldn't use modbus rtu because i had binding errors. I ran (npm install serialport) command inside server folder. Step by step:

cd ..\server
npm install serialport

than binding errors disappeared and setup a RTU connection and added variables. At this step only problem is register starting address. Some brand use 0 and others use 1 as starting address. My servo uses 0 as starting address and Fuxa uses 1 as starting address. So i couldn't access to first register.

other problems were big or little endian problem and word size. my servo had 32bit register word size. if i set fuxa variable as Int16 values at servo was messed up. of course this is normal and i set fuxa variables word sizes. i changed endian usage within servo.

after setting communication parameters appropriately i send and received values as expected. no special read or write errors i got. because i wrote values occasionally sometimes i wrote inappropriate values then i got range errors which are expected.

when i added digital input variable i got this error:
_readMemory error! Error: Modbus exception 1: Illegal function (device does not support this read/write function)

this means that servo drive doesn't support reading/writing to digital inputs or coils. this means that i need another device. (this is joke, you must laugh or not. you know). I will try with another modbus device later.

Conclusion:
there is no problem with input registers or holding registers. just setup correctly.
i will try digital inputs and coils with another device.

unocelli, 2021-07-04:
@alpaslanske Great report, thanks a lot.

unocelli, 2021-11-16:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2021-11-16 by unocelli