frangoteam/FUXA #154: _readMemory Error! & _writeMemory Error! All The Time

studuino, 2021-05-28:
Hi,
I setup several devices and in which hundreds of tags reside
I am receiving constantly _readMemory Error! and _writeMemory Error! 
How can I improve to reduce this sort of error.

Thank you.

unocelli, 2021-05-29:
Hi, thanks for the report.
Have you tried to increase the polling or activate the fragmented flag?
I'm a bit busy at the moment, I'll take care of it as soon as possible.

studuino, 2021-05-30:
Hi @unocelli 
I tried to increase polling interval to up till 5 sec but the problem didn't go away.
Only after I enabled fragmented flag with polling interval 1 second, it works like a charm.
But could you explain how fragmentation works in this case?
Thank you.

unocelli, 2021-06-01:
Hi,
Sorry for the delay. 
With fragmentation: the adjacent addresses are grouped together to split the query to read.
Without fragmentation: the read query is in one, from smaller tag address to biggest tag address (eventually split by 1000 for to big query).
Some devices do not support queries that are too large.

studuino, 2021-06-03:
@unocelli Thank you for the explanation. I understand pretty well now.
I may be able to void read/write memory error by allocating modbus address accodringly.

closed 2021-06-03 by studuino