frangoteam/FUXA #159: Program Structure for Contributing

alpaslanske, 2021-06-06:
Hi,
i want to change or add some controls. for example, i want to make some modifications on chart control.  however i don't know which file is to be changed.

is there documentation for contributing?

unocelli, 2021-06-07:
No, this is missing, do you have an idea or an example of how the documentation could be?
I admit that I prefer to focus on the features ;) I'm waiting for some help

alpaslanske, 2021-06-08:
my base idea is to know which file is to be modified when i i want to change an object. i made two changes however they are not reflected to server. as i see you are the only one working on Fuxa. I am an engineer working on Siemens and Omron PLC systems. Bad thing is i am very bad about node.js. As i mentioned my changes were not reflected to working Fuxa.

If you mention which file is to be modified if one wants to change something you an get some help. Fuxa is a promising project. I tried some open source scada programs however there were some problems with them.

For example guide me to solve problems about snap7 issue and my issue about OPC variable name problems. Let us see if i can help. Let us try to go step by step.

alpaslanske, 2021-06-08:
As a note. because i wo on Siemens and Omron PLCs, i have S71200 PLC and i can get Omron NX1 series PLC from my office.

unocelli, 2021-06-08:
You are welcome. I think it is better if you contact me on 4frango@gmail.com

closed 2021-06-08 by alpaslanske