frangoteam/FUXA #160: FSevents

alpaslanske, 2021-06-06:
Hi,

is this warning important?

npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})

unocelli, 2021-06-07:
Hi,
I think in development phase could be ignored. it would be nice to solve them, if possible ;)

alpaslanske, 2021-06-07:
Guide me. Let me try.

alpaslanske, 2021-06-08:
i searched and result is this is not an issue about Windows. Windows users can ignore this warning.

What's happening here is that a package called chokidar is being used to "watch" filesystem directories for "events" (like files being added). chokidar is a wrapper for Linux-, Windows-, and Mac-specific filesystem-watching packages, of which fsevents is the Mac variant. So,projects that uses chokidar is going to have fsevents as an optional dependency,  this WARN message can be safely ignored, as chokidar supports all common desktop architectures.

I am closing this issue.

closed 2021-06-08 by alpaslanske