frangoteam/FUXA #162: Upload image bug

wooAo, 2021-06-08:
Hi, sir
I want to upload a picture as a component
but after uploading 3 pictures,  i find it gets the same one no matter what pictures i choose
is there any bugs?

muzixiaosanshi, 2021-06-08:
Dear unocelli，
I come across the same question,Could you help to check this ?Thanks in advance.

unocelli, 2021-06-08:
Hi,
with which browser did you try? fuxa version? can you send me the output of browser console by load fuxa frontend and after adding an image?

wooAo, 2021-06-09:
Thanks for replying
I have tried chrome and firefox in local and in your live demo, it remains the same problem
I restart the project, it still dosent work
And version is v1.0.11
Here is the console ater adding 

[image]

unocelli, 2021-06-09:
I tested in the in [demo](https://frangoteam.github.io/) with png and is working.
which type of image you want to upload? SVG?

muzixiaosanshi, 2021-06-09:
I uploaded format png also and  at the beginning it can work , But if you try several times ,the problem will appear. @unocelli

wooAo, 2021-06-11:
Hi, sir
Is there a lowest browser version limit?
It fails to show up the index page, but stays in loading page and never change
Looking forward to your reply, Thanks

Here is the console and browser info
The OS is ubuntu, not windows
[image]
[image]

unocelli, 2021-06-14:
@wooAo Hi, I don't know what is the lowest browser version limit. are you sure that the problem is not the installation (nodejs and npm version)? it would be better if you opened another issue, 'Upload image bug' is not the right.

AlexVishwa, 2021-09-27:
I resolved this, try building properly and this issue won't reoccur. I reproduced the bug by running angular server and node server separately, instead one must run it as a complete system.

unocelli, 2021-09-27:
Hi,
Please send me a better description of what you mean...

AlexVishwa, 2021-09-27:
Hi Umberto,
When we try to upload an image it gives a bug of same image and new image is not selected. There is a trick that instead of building using npm start we may use nodemon main.js and vice versa which seems to clear the cache and this bug gets resolved, otherwise it is reproducing.

unocelli, 2021-09-27:
Ok, thank you,
I have to try, I thought I solved it in version 1.1.1

AlexVishwa, 2021-09-27:
No its not. It reproduces after a while.

unocelli, 2021-09-27:
Hi Alex,
it should now be fixed

AlexVishwa, 2021-09-29:
Thanks Unocelli but its not fixed. It still comes up.

AlexVishwa, 2021-09-29:
I have to restart my PC everytime. There lies an issue with clearing cache , it seems.

unocelli, 2021-10-03:
I have tried with many images and different types svg, png, jpg...
do you have any idea how to reproduce it?

AlexVishwa, 2021-10-04:
[image]
[image]
[image]
See the screenshots

unocelli, 2021-11-16:
I solved it. I will therefore close this issue, feel free to re-open it if something is still not working.

closed 2021-11-16 by unocelli