frangoteam/FUXA #165: Why do not use a professional database?

jasonlee9900, 2021-06-17:
Why do not use a professional database, e.g. mysql, postgres etc?
is that complicated to add mysql or postgres to replace DAQ?
For current DAQ database, what's the maximum limit data for usage? e.g. 1000 points/1hour ?

alpaslanske, 2021-06-18:
Maybe better however i don't like MySql. Of course this is not important. SQLite is not inferior to MySql. PostgreSQL is considerable for big projects. Is FUXA ready for big projects? I think not. There is more important issues than database.
SQLite is not limited to records. Maximum data limit depends on check interval. If your pullrate is 1000ms then you acquire 60 points in a minute, 3600 points in an hour. 86400 points per day. That depends on you and your system. If you decrease pullrate to 500ms than you acquire 86400x2 points per day.

jasonlee9900, 2021-06-18:
I installed FUXA, it's a fantastic SCADA system, the best open source SCADA project in recent year! I mean the design structure, although I haven't test it's stability. Nearly everything is there, It seems ready for big project now. 
Only thing I worry about is database, most of my projects required 300~600 points/every 3 minutes, so it will 288,000 points each day, and we hope data can be save for 5~10 years, so it will be 525,600,000‬~1,051,200,000 points, can sqllite still run smoothly on history curve etc. function? so I suggest you add postgres as second database. 
BTW, may you add a login page? and report PDF, csv, excel etc.？

unocelli, 2021-06-18:
@jasonlee9900 
Hi,
thank you for your appreciation. In the origin FUXA was born as process visualization so relatively few variables to save. I'm agree with you that a scada software must be able to store a large amount of data in a high-performance way for the history and sqlite has its limits. hooking up a more powerful database is a feature I would like to address. I think the discussion on the choice of database is also open:  MySql, PostgreSQL, MongoDB, InfluxDB #32, ...

concept10, 2021-06-18:
My idea is to have a pluggable transport allowing any database to be used.  Customer choice maybe?

jasonlee9900, 2021-06-19:
> My idea is to have a pluggable transport allowing any database to be used. Customer choice maybe?

Most of IoT systems are designed based on Node-red, as FUXA is also based on node.js, so I wonder whether FUXA is also based on Node-Red?  If so, could FUXA open it so that we can add node-red functional mod by our-self.

Talbourdet, 2021-07-10:
I think it would be interesting to make possible the integration of database driver by plugin because there are plenty of possible databases

Harshid93, 2021-07-31:
it's Very Much Useful if you can Add Database facility , I have Developed seperate software for Database but if it is integrated with FUxa it will be more usefull tool for us.

unocelli, 2022-09-08:
going to close this as resolved. let me know if you have any issues.

closed 2022-09-08 by unocelli