frangoteam/FUXA #168: Is there any plan to extension the logic programing（js) like node red.

loringao, 2021-06-23:
This is a amazing project , Thanks fuxa team.

We have a scenarios, it is a line control project(50%data collecton and display, 50%logic process with MES):
1. collect the data use fuxa
2. do some logic in background
3. show the real time/calculated data in HMI

We investigated red-node before, it's SCADA and configuration designer is not enough for user experience.  do we have plan to support the logic extension?

unocelli, 2021-06-23:
Hi,
yes, it is in the TODO list, but the time is missing :(
The idea is to have in frontend like a code editor for the javascript script (with special tags to bind I/O device), a settings to configure the scheduler to launch scripts (per loop interval, date, time) and in backend a component to manage and run the scripts (logic).

loringao, 2021-06-24:
Got it, Thanks.

closed 2021-06-24 by loringao