frangoteam/FUXA #174: Map Control

eyadalhanafi, 2021-07-09:
Hi, Thank you for your efforts.
I installed and tested FUXA, it looks a very promising Open Source SCADA system. I have checked many open source SCADA systems, I found Fuxa one of the best in terms of the structure and ease of use.
As a new feature enhancement, can you add Map control that displays the status and parameters of locations control system on an interactive map? The map allows to monitor geographically control systems and navigate to the details page of a location.

unocelli, 2021-08-03:
Hi, sorry for the delay.
I agree it is definitely a great feature. But if I'm not mistaken, it is dependent to subscription costs to access the to the maps.

eyadalhanafi, 2021-08-05:
Hi,

I'm not sure if Leaflet is suitable for map control.  https://github.com/Leaflet/Leaflet.

An extension to Leaflet that contains configurations for free tile providers which is needed.
https://github.com/leaflet-extras/leaflet-providers

unocelli, 2021-08-08:
Hi, 
very interesting, I love it. if we don't find someone who is interested to implement it in fuxa, I will try as soon as possible.

Thanks

killerbeat12, 2024-06-12:
Maybe to keep it easy, implementing just a dialog box that is opening google maps and giving the lattitude and longitude parameters with it should already be a nice and easy solution?

unocelli, 2025-03-04:
I’m going to close this as resolved. let me know if you have any issues.

closed 2025-03-04 by unocelli