frangoteam/FUXA #175: I/O Communication and Memory Tags

eyadalhanafi, 2021-07-09:
In general and up to my knowledge, there are two different categories of tags in SCADA systems: I/O communication tags and memory tags. 

1 - I/O tags- These include single points and multi-points. Points have device addresses that are mapped to some memory location in the controller device. Those tags can be defined in Fuxa 

2 - Internal Memory Tags (they do not have I/O addresses and are used solely in the SCADA system. ) - These include 
	2.1 Constant points; Global Internal Tags. Constant Tags are most frequently used for operator entry of values.
	2.2 Accumulation Points; consists of an Integration function from another I/O tag. Used for Totalizers (Example: Calculate volume from flow rate)
	2.3 Calculation Points; Used to  perform mathematical and logical operations on the values of other tags.  functions include addition, Subtraction, Multiplication, Division, Sine function, etc. The result of calculation (calculation point) is a single tag that can be logged and alarmed.
	2,4 System points. System Point tags provide system diagnostics and other system related info. System "datetime" can be defined as a system tag. 

I think it is important to consider other memory tags in Fuxa.

unocelli, 2021-08-03:
Hi, sorry for the delay.
I am not sure if I have understood your remarks. FUXA displays tag values and modifies them if necessary. You decide which tags, whether they are I/O's or internal variables is irrelevant. When I think of devices communicating via opcua, you don't see the difference.

eyadalhanafi, 2021-08-04:
No issue.
I noticed there is a driver "internal" for memory tags. that is really very good but i think it is not enough. I think other types of tags should be maintained.

For example, if I need to do math operation like multiplication or division for an I/O tag,  we usually create new tag called "calculated tag" to have the result saved there.

Another example; if i want to count how many times Boolean tag become "ON" like Pump start, there should be a mechanism like a formula to do that. The easiest way is to have "accumulated tag" to keep that value in.

maybe there is another way to handle that like keep that on PLC or RTU level as part of the PLC/RTU program but usually SCADA systems have a mechanism to handle those type of formulas or calculated tags.  

hope I made myself clear.

unocelli, 2021-08-04:
would be in my plans to give the possibility to create script for own logic that run in server side. I think this would also apply to this problem.

unocelli, 2022-05-08:
Let me know if something doesn't work as expected

closed 2022-05-08 by unocelli