frangoteam/FUXA #176: This project cannot use reverse proxy.

Drincann, 2021-07-15:
For example:

I start the FUXA service at localhost:1881.

```js
const app = require('express')();
const proxy = require('express-http-proxy')

app.use('/proxy', proxy('http://localhost:1881'));

app.listen(80);
```
I visit

http://localhost/proxy

http://localhost/proxy/

http://localhost/proxy/home

http://localhost/proxy/home/

http://localhost/proxy/home/index.html

[image]

[image]

Static resource files seem to be accidentally loaded from absolute paths.

unocelli, 2021-08-03:
Hi, sorry for the delay.
I don't know if I can help you, I've never tried with a proxy...we have to investigate.

unocelli, 2022-10-09:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2022-10-09 by unocelli

children1987, 2025-07-07:
btw, I'm new in nodejs and angular.  Thank you so much for your perfect codes! @unocelli