frangoteam/FUXA #177: modbusTCP 5 digits to 6 digits

anhthe, 2021-07-26:
I am interested in FUXA, and i have a problem with modbusTCP.
I created new tag with address is 400001 and map with 40001 in modbus tool as pitures below:
[image]
[image]
 But I got a error:  'modbus' _readMemory error! Error: Modbus exception 2: Illegal data address (register not supported by device)

Please show me what is the error?

Thank you!

alpaslanske, 2021-07-26:
that is not about Fuxa. Try this settings.
[image]

anhthe, 2021-07-27:
> that is not about Fuxa. Try this settings.
> [image]

Thank you Alpaslanske!

closed 2021-07-27 by anhthe