frangoteam/FUXA #178: SNMP driver

eyadalhanafi, 2021-07-29:
I would like to know if SNMP driver is in the road map of future development.
Thank you

unocelli, 2021-08-03:
Hi,
why not...we need to find some interesting javascript library for the backend

eyadalhanafi, 2021-08-04:
Hi,

I'm not expert in JavaScript but i found something that could be used. You are better than me to decide.
https://github.com/markabrahams/node-net-snmp#readme

it is JavaScript implementation of the Simple Network Management Protocol (SNMP) - full functionalities are included in this library.

unocelli, 2021-08-08:
Hi, interesting.
it's not on my immediate plan to extend the devices protocols, I would be very interested in a pull request if you or anyone else is inclined to do this.