frangoteam/FUXA #180: Trying to get in touch regarding a security issue

zidingz, 2021-08-02:
Hi there,

I couldn't find a `SECURITY.md` in your repository and am not sure how to best contact you privately to disclose a security issue.

Can you add a `SECURITY.md` file with an e-mail to your repository, so that our system can send you the vulnerability details? GitHub suggests that a [security policy](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) is the best way to make sure security issues are responsibly disclosed.

Once you've done that, you should receive an e-mail within the next hour with more info.

Thanks! (cc @huntr-helper)

unocelli, 2021-08-03:
Hi,
thanks for the suggestion.

JamieSlome, 2022-04-28:
@unocelli - apologies for letting this slip through. We sent some e-mails to your organisation's e-mail address but didn't get any response. You can find the reports here:

https://huntr.dev/bounties/942176f4-c1c9-45a0-a237-b567b8765061/
https://huntr.dev/bounties/bf6ec194-1592-4ca5-957e-318bb27c584b/
https://huntr.dev/bounties/fc578b92-642c-4dfe-91cc-99f7a2b4c4ef/
https://huntr.dev/bounties/dda19a53-e9ee-4bee-b232-4ca7895ed3a4/
https://huntr.dev/bounties/d4fe6bc9-46ed-4df9-ac1e-315e2f684d8b/

They are all private and only accessible to maintainers with repository write permissions 👍 

(cc) @aravindd007 @h3v0x @galapag0s

unocelli, 2022-04-28:
Hi Thanks a lot for the reports.
I will of course look at them as soon as possible, something I have already fixed

JamieSlome, 2022-04-29:
@unocelli - great, let me know if you have any questions ❤️

closed 2023-12-30 by unocelli