frangoteam/FUXA #182: Alarms

eyadalhanafi, 2021-08-04:
Hi. Thank you for your support.
I defined "internal" communication device with internal memory tag for testing alarms. I setup the alarms HiHi and Hi using that memory tag.  I defined alarms ranges as needed. when i inject value in the memory tag that in the range of HiHi or Hi alarm to generate an alarm, No alarms generated. 
I don't know if i did something wrong so that no alarms generated.  
does Alarms work with "internal" device driver?

unocelli, 2021-08-04:
Hi,
I'm sorry, internal is not currently managed by the server, it is used for context variables on the frontend...I have to extend it. The idea would be to use it in the server side, in particular to create 'memory' variables (#175)  and also with the possibility to write some logic script, and so also to generate alarms.

unocelli, 2022-05-08:
going to close this as resolved. let me know if you have any issues.

closed 2022-05-08 by unocelli