frangoteam/FUXA #188: save multiple projects

YasithAri, 2021-08-18:
Hi, i am checking the FUXA for a project and seems to be very interesting. have a requirement that create multiple projects on same server and same db , is that possible? for ex. user A logging to the FUXA and create a project and also user B creates a project. so both should save in db and access accordingly. as i understand it override the same project with new one. please advise

unocelli, 2021-08-21:
Hi,
I don't understand your goal? 
it is possible to start multiple fuxa instances by defining different ports.

YasithAri, 2021-08-22:
if we set the different ports , will it be share the same app db and other d files? my goal is one user has set of api devices and other user have serperate devices . so if i use one port for both users, can db handle both data at same time?

unocelli, 2023-02-04:
we analysed the feature and realised that require a redesign that would need many resources.

closed 2023-02-04 by unocelli