frangoteam/FUXA #189: Modbus TCP Floating Point

eyadalhanafi, 2021-08-19:
Thank you for your support.
We are using Modbus TCP driver to connect to a PLC and poll some floating points. The communication was established fine and we could receive the data but the data received are integers, not deiciml numbers. I checked valuse using MODSCAN and the data received are with decimals. 
[image]

I don't know if i missed anything in the configuration so that i received that. Please advise.

unocelli, 2021-08-22:
Hi,
you are looking at the right address? which value you receive if you address to 402565 ?

eyadalhanafi, 2021-08-22:
Hi,
The address 402565 gives wrong value.

unocelli, 2021-08-22:
Hi,
Interesting is that if I set the value from FUXA or from the modscan32 I see the correct value
[image]

eyadalhanafi, 2021-08-23:
Hi
I tried another SCADA system with modbus driver, I received the correct value. It is weired

eyadalhanafi, 2021-08-28:
I tried to write on floating point signal from FUXA,  I didn't get the value in Modscan. I received incorrect value.

unocelli, 2021-09-19:
I added the LittleEndian type, maybe this was the problem?

unocelli, 2021-11-16:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2021-11-16 by unocelli