frangoteam/FUXA #191: Modbus - Data length error, expected 263 got 7

eyadalhanafi, 2021-08-25:
Hi, 
We got an error in modbus TCP driver. We defined some signals boolean & holding registers. At certain point and after defining new holding register signal, we recived an error .

[error] 'PLC' _readMemory error! Error: Data length error, expected 263 got 7

[image]

your support is appreciated.

unocelli, 2021-08-25:
Hi, thanks for the report.
Have you tried to increase the polling or activate the fragmented flag?

eyadalhanafi, 2021-08-28:
Currently the PLC is in production and another SCADA system is connected with the PLC device with polling every 1 second without any issue. 
Currently we are testing FUXA with production system juts to evaluate the performance of the system when we have too many signals and update rate every 1 second.

unocelli, 2021-08-29:
Hi, thanks for your report.
I'm not sure of there are some issue by the node-modbus-serial library. I'm making some tests too and I see that there are problems with the buffer size in the communication by ask to big tags array.
currently I split the tags array size to 1000 for section. can you try to set the limit to 100 and let me know if there are any improvements? thanks.

The value to change is in '/server/runtime/devices/modbus/index.js' line 8, and then restart the server
`const TOKEN_LIMIT = 100;`

eyadalhanafi, 2021-08-31:
We changed the TOKEN_LIMIT as per your instructions but we did not see any changes or improvements. Same warning messages in the log file. We tried Poll time less than 5 seconds, unfortunately when we did so, we started receiving errors and signals readings did not received.

unocelli, 2021-08-31:
Try TOKEN_LIMIT=100 and without fragmentation...which plc are you using?

eyadalhanafi, 2021-09-01:
Hi, thank you for your support.
We disabled fragmentation and set TOKEN_LIMIT=100 and Poll 5 second, No errors or polling overload.  When changed the poll time less than 5 seconds, we started getting errors.

Regarding PLC, we are using APAX controller APAX-5620 from Advantech https://www.advantech.com/products/1-cyxw62/apax-5620kw/mod_80991767-1639-48f4-8b2e-0660e350c79b

unocelli, 2021-09-12:
Hi, with the last version I changed the delay betwen the frame to 5 milliseconds. I tested with ModRSsim2 as device and 600 tags (200 boolean, 200 integer and 200 float) and 1 second polling without any issue.

unocelli, 2021-11-30:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2021-11-30 by unocelli