frangoteam/FUXA #200: Modbus plugin instllation not work

adolfosan, 2021-10-06:
When installing the modbus plugin it shows me the message: "Installing ... OK!", but when I go to select a Modbus device it is not found in the fuxa server connections, in the terminal it shows me the caution message: "try to create d but plugin is missing! "
[image]
[image]

adolfosan, 2021-10-06:
I am using npm, the version is 1.1.1

adolfosan, 2021-10-06:
I installed with first option: npm  and I'm using 1.1.1 version

unocelli, 2021-10-06:
I'm sorry is there a bug in css you have to click in the corner
[image]

unocelli, 2021-10-06:
I have already fixed but not release in a version. Coming soon.
you still have the problem with the modbus plugin?

adolfosan, 2021-10-06:
> you still have the problem with the modbus plugin?

Yes

adolfosan, 2021-10-06:
> look good...I just tested on ubuntu 20 without any problems... you can try to install your self [modbus](https://www.npmjs.com/package/modbus-serial):
> 
> > cd server
> > npm i --save modbus-serial

Ok, let me try, I have a question:
Cuando yo instale modbus-serial , saldra en la creacion de un dispisitivo los tipos:,Modbus RTU,Modbus TCP?

closed 2021-12-20 by unocelli