frangoteam/FUXA #205: Unexpected error

MRIIOT, 2021-10-10:
Not reproducible yet.

```
root@5368cd5e2b6f:/usr/src/app/FUXA/server# cat /root/.npm/_logs/2021-10-10T15_31_43_711Z-debug.log
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@6.14.15
3 info using node@v14.18.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle fuxa-server@1.1.2-489~prestart: fuxa-server@1.1.2-489
6 info lifecycle fuxa-server@1.1.2-489~start: fuxa-server@1.1.2-489
7 verbose lifecycle fuxa-server@1.1.2-489~start: unsafe-perm in lifecycle true
8 verbose lifecycle fuxa-server@1.1.2-489~start: PATH: /usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/usr/src/app/FUXA/server/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle fuxa-server@1.1.2-489~start: CWD: /usr/src/app/FUXA/server
10 silly lifecycle fuxa-server@1.1.2-489~start: Args: [ '-c', 'node main.js' ]
11 silly lifecycle fuxa-server@1.1.2-489~start: Returned: code: 137  signal: null
12 info lifecycle fuxa-server@1.1.2-489~start: Failed to exec start script
13 verbose stack Error: fuxa-server@1.1.2-489 start: `node main.js`
13 verbose stack Exit status 137
13 verbose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:400:28)
13 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:400:28)
13 verbose stack     at maybeClose (internal/child_process.js:1058:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:293:5)
14 verbose pkgid fuxa-server@1.1.2-489
15 verbose cwd /usr/src/app/FUXA/server
16 verbose Linux 5.10.52-v7l+
17 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
18 verbose node v14.18.0
19 verbose npm  v6.14.15
20 error code ELIFECYCLE
21 error errno 137
22 error fuxa-server@1.1.2-489 start: `node main.js`
22 error Exit status 137
23 error Failed at the fuxa-server@1.1.2-489 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 137, true ]
root@5368cd5e2b6f:/usr/src/app/FUXA/server# 
```

MRIIOT, 2021-10-11:
Here is another one.

```
    at Database.<anonymous> (/usr/src/app/FUXA/server/node_modules/sqlite3/lib/sqlite3.js:18:25)
,    at /usr/src/app/FUXA/server/runtime/storage/daqnode.js:541:23
,    at new Promise (<anonymous>)
,    at _insertTagToMap (/usr/src/app/FUXA/server/runtime/storage/daqnode.js:539:16)
,    at MQTTclient.DaqNode.addDaqValues [as addDaq] (/usr/src/app/FUXA/server/runtime/storage/daqnode.js:191:44)
,    at MQTTclient.polling (/usr/src/app/FUXA/server/runtime/devices/mqtt/index.js:144:34)
,    at Device.polling (/usr/src/app/FUXA/server/runtime/devices/device.js:141:14)
,    at Timeout._onTimeout (/usr/src/app/FUXA/server/runtime/devices/device.js:156:22)
,    at listOnTimeout (internal/timers.js:557:17)
,    at processTimers (internal/timers.js:500:7)
,--> in Database#run("INSERT INTO data (id, name, type) VALUES('t_6dee61b7-a3614793','ff[a]','json'); SELECT last_insert_rowid() FROM data", [Function (anonymous)])
,    at /usr/src/app/FUXA/server/runtime/storage/daqnode.js:541:23
,    at new Promise (<anonymous>)
,    at _insertTagToMap (/usr/src/app/FUXA/server/runtime/storage/daqnode.js:539:16)
,    at MQTTclient.DaqNode.addDaqValues [as addDaq] (/usr/src/app/FUXA/server/runtime/storage/daqnode.js:191:44)
,    at MQTTclient.polling (/usr/src/app/FUXA/server/runtime/devices/mqtt/index.js:144:34)
,    at Device.polling (/usr/src/app/FUXA/server/runtime/devices/device.js:141:14)
,    at Timeout._onTimeout (/usr/src/app/FUXA/server/runtime/devices/device.js:156:22)
,    at listOnTimeout (internal/timers.js:557:17)
,    at processTimers (internal/timers.js:500:7)
,2021-10-10T15:44:56.123Z [ERR] 	daqstorage.add-daq-values addDaqfnc failed! 'd_a07a70c6-c4ec486a' undefined
,(node:19) UnhandledPromiseRejectionWarning: ReferenceError: _checkWorking is not defined
,    at /usr/src/app/FUXA/server/runtime/storage/daqnode.js:214:25
,(node:19) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 152)
,
,> fuxa-server@1.1.2-489 start /usr/src/app/FUXA/server
,> node main.js
,
,2021-10-10T15:45:52.404Z [INF] 	FUXA V.1.1.2-489
,2021-10-10T15:45:53.539Z [INF] 	FUXA init in  1194ms.
,2021-10-10T15:45:53.578Z [INF] 	'mqtt' created
,2021-10-10T15:45:53.590Z [INF] 	'mqtt' start
,2021-10-10T15:45:53.598Z [INF] 	FUXA started!
,2021-10-10T15:45:53.610Z [INF] 	WebServer is running http://127.0.0.1:1881/
,2021-10-10T15:45:53.835Z [INF] 	'mqtt' connected!
,2021-10-10T15:45:57.227Z [INF] 	socket.io client connected
,GET /api/settings 304 9.246 ms - -
,GET /api/settings 304 2.281 ms - -
,GET /api/settings 304 1.465 ms - -
,GET /api/project 304 43.661 ms - -
,2021-10-10T15:46:05.972Z [INF] 	socket.io client connected
,POST /api/projectData 200 53.738 ms - -
,GET /api/settings 304 2.626 ms - -
,GET /api/settings 304 2.441 ms - -
,GET /api/settings 304 2.528 ms - -
,2021-10-10T15:46:13.117Z [INF] 	socket.io client connected
,GET /api/project 200 25.424 ms - 320644
,POST /api/projectData 200 54.176 ms - -
,GET /api/settings 304 7.001 ms - -
,GET /api/settings 304 1.379 ms - -
,GET /api/settings 304 1.803 ms - -
,2021-10-10T15:46:42.212Z [INF] 	socket.io client connected
,GET /api/project 200 22.884 ms - 320644
,2021-10-10T15:47:19.021Z [INF] 	socket.io client connected
,2021-10-10T15:47:47.596Z [INF] 	socket.io client connected
,2021-10-10T15:48:46.976Z [INF] 	socket.io client connected
,2021-10-10T15:49:14.973Z [INF] 	socket.io client connected
,2021-10-10T15:49:58.024Z [INF] 	socket.io client connected
,2021-10-10T15:50:44.024Z [INF] 	socket.io client connected
,2021-10-10T15:51:26.976Z [INF] 	socket.io client connected
,2021-10-10T15:52:12.084Z [INF] 	socket.io client connected
,2021-10-10T15:52:56.483Z [INF] 	socket.io client connected
,2021-10-10T15:53:29.008Z [INF] 	socket.io client connected
,2021-10-10T15:54:25.063Z [INF] 	socket.io client connected
,2021-10-10T15:54:47.937Z [INF] 	socket.io client connected
,2021-10-10T15:55:52.396Z [INF] 	socket.io client connected
,2021-10-10T15:56:21.159Z [INF] 	socket.io client connected
[... 209 lines cut ...]
,    at /usr/src/app/FUXA/server/runtime/storage/daqnode.js:541:23
,    at new Promise (<anonymous>)
,    at _insertTagToMap (/usr/src/app/FUXA/server/runtime/storage/daqnode.js:539:16)
,    at MQTTclient.DaqNode.addDaqValues [as addDaq] (/usr/src/app/FUXA/server/runtime/storage/daqnode.js:191:44)
,    at MQTTclient.polling (/usr/src/app/FUXA/server/runtime/devices/mqtt/index.js:144:34)
,    at Device.polling (/usr/src/app/FUXA/server/runtime/devices/device.js:141:14)
,    at Timeout._onTimeout (/usr/src/app/FUXA/server/runtime/devices/device.js:156:22)
,    at listOnTimeout (internal/timers.js:557:17)
,    at processTimers (internal/timers.js:500:7)
,--> in Database#run("INSERT INTO data (id, name, type) VALUES('t_a48eb05e-05044acf','fuxa/job[active]','json'); SELECT last_insert_rowid() FROM data", [Function (anonymous)])
,    at /usr/src/app/FUXA/server/runtime/storage/daqnode.js:541:23
,    at new Promise (<anonymous>)
,    at _insertTagToMap (/usr/src/app/FUXA/server/runtime/storage/daqnode.js:539:16)
[... cut ...]
,    at processTicksAndRejections (internal/process/task_queues.js:95:5)
,(node:20) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 36)
```

unocelli, 2022-10-09:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2022-10-09 by unocelli