frangoteam/FUXA #206: MQTT Dynamic JSON Payload Handling

MRIIOT, 2021-10-10:
1. Create subscription based on this payload: 

```
{
  "a" : 5,
  "b" : 900
}
```
2. Bind controls to 'a' and 'b'

3. Send this payload:

```
{
  "a" : 5
}
```

4. Control bound to 'b' displays the entire JSON string.

Expected behavior:
* Update controls bound to 'b' to: undefined or null
* Do not update controls bound to 'b'.  Leave as previous value (eg. 900)

[image]

[image]

[image]

unocelli, 2021-10-10:
Hi, thanks for the report.
I would prefer the second option:
- Do not update controls bound to 'b'. Leave as previous value (eg. 900)

MRIIOT, 2021-10-10:
> Hi, thanks for the report. I would prefer the second option:
> 
> * Do not update controls bound to 'b'. Leave as previous value (eg. 900)

I agree.

unocelli, 2021-12-20:
going to close this as resolved. let me know if you have any issues.

closed 2021-12-20 by unocelli