frangoteam/FUXA #226: Duplicated shape animation

eyadalhanafi, 2021-11-24:
Hi,
If we duplicate a shape with animation, the duplicated shape doesn't have animation.

unocelli, 2021-11-24:
Hi,
you mean that the actions values aren't copied? how you duplicate the shape with actions?

eyadalhanafi, 2021-11-25:
I mean color change fill property. I tried both "duplicate element" and "copy & paste"

unocelli, 2021-11-25:
[image]

eyadalhanafi, 2021-11-30:
Thank you for your efforts.
I meant by that general shape like rectangle. I configured fill property for that shape. Same shape duplicated, the fill property is not propagated to the duplicated shape.

unocelli, 2021-11-30:
You mean the rectangle of **General** and not of **Shape**, I can reproduce it and will fix it as soon as possible.
Thanks

unocelli, 2023-02-04:
I am going to close this as resolved. let me know if you have any issues.

closed 2023-02-04 by unocelli