frangoteam/FUXA #231: flex-variable component is not different with demo site

echolove38, 2021-11-30:
[image]
[image]
look at these two pics, the first( code downloaded from your github) have no device binding ui,
are you sure the github code is the newest code ?

unocelli, 2021-11-30:
Hi, yes you're right the **demo is an old version** that I still have to update it. in any case, the current version is on the master branch

echolove38, 2021-12-01:
> Hi, yes you're right the **demo is an old version** that I still have to update it. in any case, the current version is on the master branch

will you provide a uncompressed fuxa-editor.js，I want to extend something When need

unocelli, 2022-10-09:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2022-10-09 by unocelli