frangoteam/FUXA #234: Mulitpile device use same view

s001mtzniot, 2021-12-10:
Hi:
   This is a powerfaul solution, for embed to other it system, how can i set one view to reused for all same sensor, for examle, i create one view for one device ie. temputre A, But how user select device B(also is a tempure sensor) how to reuse the view?
thanks

unocelli, 2021-12-10:
Hi,
You have to define a device as 'internal' and some variable (Tag) and bind it to the controls of reusable 'View' and than with events click to show the 'View' as 'Page', 'Dialog', 'Card' add context variable to bind: internal variable to device tag...I want to add an explanation in wiki as soon as possible.

eyadalhanafi, 2022-01-04:
Further to the same point, we have 20 valves. We created one "Card" (faceplate) and define internal variables to bind with the actual valve tags. We are wonder how can we map Internal variable to none device Tag like Valve Name. Each card has a label for the Valve ID and some other texts that are unique for each valve.  Those changed based on the valve selection
Appreciate your efforts.

unocelli, 2022-01-04:
Let me try to explain:
I defined a valves view and a valve dialog to configure
[image]
[image]

I bind the 2 valves value to 2 Modbus tag
[image]

I defined a internal 'device' with a variable (tag) for the title dialog and a variable for the input value and i bind the controls
[image]

I defined in Valves view the 2 valve shape the 'Events': 
- click to open the card with context variable and bonded the internal variable to the Modbus tag
- click to set value for the dialog title
[image]

You can define a dialog button to confirm the input value and close the dialog
[image]

You can use with key ENTER or with confirm button
[image]

eyadalhanafi, 2022-01-04:
Thank you so much.

closed 2022-02-28 by unocelli