frangoteam/FUXA #237: feature request: pipe editor improvements

burner-, 2021-12-13:
Currently pipe editor will make by default huge pipes. Those sliders make modifying quite hard. Looks that it is some how buggy so that pipe looks ok at first time but if I open parameters and close them it will make pipes really fat. 
So I suggest three modifications:
- Change sliders to numeric up & downs. It helps editing a lot. 
- Add default pipe parameters to project settings. 
- fix defaluts so that property editor wont blow pipe to super fat at first edit.

unocelli, 2021-12-20:
Hi, thank you for your suggestions, I agree with you and will make the changes as soon as possible ;)

unocelli, 2022-01-06:
going to close this as resolved. let me know if you have any issues.

closed 2022-01-06 by unocelli