frangoteam/FUXA #238: AB Ethernet I/P

cjmccoy2, 2021-12-19:
Has support been added for AB Ethernet IP devices? I see a couple issues on it. 

If so, How do I add it to my project as it is not an available device

unocelli, 2021-12-20:
Hi, you mean ABB plc? have you already tried with EthernetIP device?

cjmccoy2, 2021-12-20:
AB, Allen Bradley PLC, Ethernetip is not a device option for me

unocelli, 2021-12-20:
If it's because you don't see it, you have to install it in the 'Plugins' section.
[image]

unocelli, 2021-12-20:
We have tested #118 and was not so bad

cjmccoy2, 2021-12-20:
Okay, I must be missing something. This is what my plugin section looks like
[image: image.png]

unocelli, 2021-12-20:
I don't see the image :(  which version did you installed?

cjmccoy2, 2021-12-20:
[image]

Should be 1.1.3

unocelli, 2021-12-20:
is not currently in a release. but if you download the last version from [master](https://github.com/frangoteam/FUXA/archive/refs/heads/master.zip) you find it (the same in docker frangoteam/fuxa:latest)

cjmccoy2, 2021-12-20:
Thank your for the help, I am pretty knew to all of this. I've got the drivers added but cant get any of the values to update. I have tried with 2 PLC's. Micro 850 and MicroLogix 1100. Both are ethernetip. All tags are global. Is there a differnt way to format the tag names?

[image]
[image]

unocelli, 2021-12-20:
I can't help you because I don't use it. maybe are the address wrong, take a look to the library documentation that we use  [nodepccc](https://github.com/plcpeople/nodepccc)
you can try to contact [Alex](https://github.com/AlexCastro-Prog) he helped me to integrate the library and has good experience of the field #118

unocelli, 2021-12-26:
As info, you can try to look how the nodepccc library are used in node-red application, maybe you find more information...

cjmccoy2, 2021-12-27:
Would you be able to implement this driver? It supports the Micro800 line of PLC’s. Allen Bradley also had a free simulator that it could be tested with.

https://github.com/msvargas/node-logix

AlexCastro-Prog, 2022-01-04:
Hello, guys take a look at my latest video related to Rockwell plc's to connect with FUXA Web Scada.

https://youtu.be/iTz7ryccLWw

Please subscribe, comment and give your like!!!!

Thanks!

Auyerix, 2022-01-19:
Excelente video a great collaboration... FUXA is awesome... I will also test it with Rockwell, I'm currently working with MQTT and works fine!!! Good Job!!!

Nug0, 2022-04-01:
> 

hello,

i watched your video. When your entering your port how did you determin which to use? im trying to connect to a PLC i can ping it and its on a local network just cant get FUXA to talk to it. Also with compact logix normally the adress needs a /0/1 at the end, how do i do that with the routing?

ZhuNathan, 2022-08-23:
> I can't help you because I don't use it. maybe are the address wrong, take a look to the library documentation that we use [nodepccc](https://github.com/plcpeople/nodepccc) you can try to contact [...]
Alex show us the way to read/write data file in Control/CompactLogix , it's necessary to built a data file map in PLC, in order to link the tag arrays, it worked for sure, that's a good job, but it's might be more useful and popular to read/write Control/CompactLogix tags directly instead of making a mapping and downloading again.......  I think it might be good to separate the driver "EthernetIP" to "ControLogixTCP" and "SLC500TCP", or any other labels to differentiate Contol/CompactLogix and SCL500/Miroc 800 families, it might be a BIG modification I know. I can build a CompactLogix platform for you to test your code, if it's necessary.

ZhuNathan, 2022-08-23:
> I can't help you because I don't use it. maybe are the address wrong, take a look to the library documentation that we use [nodepccc](https://github.com/plcpeople/nodepccc) you can try to contact [...]

Actually, the EthernetIP in FUXA is only support PCCC communication like SLC500/Micro 800 family PLC or a data file mapping, the Control/CompactLogix is another communication called CIP, I think is necessary to separate the driver "EthernetIP" to "ControlLogixTCP" and "SLC500TCP", or any other labels to differentiate the Control/CompactLogix and SLC500/Micro800 families, it would be a BIG modification I know, but it would be more useful and popular in future.