frangoteam/FUXA #239: Scalling problem in shapes and controllers

patsonkaushik, 2021-12-24:
I want to scale in component and shape but it's not working with given svg .. can you guide me how to solve this issue???

unocelli, 2021-12-24:
Hi, I do not understand what you mean. can you give me an example?

patsonkaushik, 2021-12-25:
I am going to do like this video in fuxa editor but getting errors.
please find attachment video ..
https://user-images.githubusercontent.com/77775964/147378331-96510b51-4845-48dc-9af2-ecf5d4285fb2.mp4

unocelli, 2021-12-26:
Currently you can hide and show a shape, if you will implement the scale effect and you will that I try to help you, you have to share your code

patsonkaushik, 2021-12-27:
I have made like this:
=> when user select "scale " option , he will get 2 sub option how to scale selected object.
it has 2 option 1) first screenshot is scaling is by height or width or both
2) 2nd screenshot is about when user select a) top-left it will zoom out from top left likewise next options...
[image]
[image]
[FUXA-master3.zip](https://github.com/frangoteam/FUXA/files/7778383/FUXA-master3.zip)

Also consider this video ..
https://user-images.githubusercontent.com/77775964/147378331-96510b51-4845-48dc-9af2-ecf5d4285fb2.mp4

unocelli, 2022-01-02:
Hi,
you don't have to check the selection of optionsScale. 
you have to remove from flex-action.component.html file the (selectionChange)="onCheckActionScaleType(item.type, item)" 
The error occurs because you don't have defined the onCheckActionScaleType function

unocelli, 2022-01-02:
you must also implement the scale effect in 'processAction' by shapes.component.ts

patsonkaushik, 2022-01-03:
I have defined onActioscaletype function now errors is gone but i am going to implement scale effects .. how to implement it with SVG components...
find attachment . i have uploaded latest code here..
[FUXA-master4.zip](https://github.com/frangoteam/FUXA/files/7800594/FUXA-master4.zip)

unocelli, 2022-01-03:
You have to look by [svg library](https://svgjs.dev/docs/3.0/manipulating/#scale) and manage the options consequently in 'processAction' by shapes.component.ts

```
static processAction(act: GaugeAction, svgele: any, value: any,  gaugeStatus: GaugeStatus) {
   ...
         } else if (this.actionsType[act.type] === this.actionsType.scale) {
            if (act.range.min <= value && act.range.max >= value) {
                let element = SVG.adopt(svgele.node);
                element.scale(2);    //, 0, 0);
            }
        }
    }
```
[shapes.component.zip](https://github.com/frangoteam/FUXA/files/7800812/shapes.component.zip)

AndBat86, 2025-06-10:
Hello and sorry for the question, but I'm new with FUXA and I searched a lot without any success. I can't find the action to scale my shapes. Is this action enabled only for svg shapes?
My goal is to build a minimal HMI that must be used in smartphone and tablet. I need a minimum responsiveness. Pages don't resize automatically to fit the screen and the only way I foud is working with scale action. Is this possible? Is there a better way to archive my goal?