frangoteam/FUXA #240: Import a list of tags Feature

cjmccoy2, 2021-12-24:
It would be great to import a list of tags from excel. This would make this process much more efficient.

Device, Tag Name, Address
[image]

Great work so far!

unocelli, 2021-12-26:
hi, I think it is a good idea, but from a csv file and not from excel

unocelli, 2022-06-17:
Hi,
I added the feature, let me know if you have any issues.

Kortenbach, 2022-06-17:
Wow, that's a great addition! Will look into it next week.
BTW, (off topic) is there some way to make a graph of measurements that are like 4 weeks old? Can Grafana be used with FUXA?

unocelli, 2022-07-26:
going to close this as resolved. let me know if you have any issues.

closed 2022-07-26 by unocelli