frangoteam/FUXA #242: Cant browse mqtt topics in latest version

thingstanje, 2022-01-04:
Today i run the fuxa docker edition with latest edition. Doing the same setup for my mqtt driver, I cant browse the topics like i did last friday on version v1.1.3. Its working perfectly in v1.1.3 docker edition

How to pull the docker edition of v.1.1.3?

thingstanje, 2022-01-04:
[image]

thingstanje, 2022-01-04:
The above image is appeared when i use v.1.1.3

thingstanje, 2022-01-04:
> The above image is appeared when i use v.1.1.3

unocelli, 2022-01-04:
I have just tested it without any issues.
you have some error in browser console or in server logs?

thingstanje, 2022-01-07:
Problem was due to my internet connection. sorry for inconvenience

thingstanje, 2022-01-07:
Dear unocelli,
I checked again this morning. Unfortunately I cant browse the mqtt topics again. No error displayed from fuxa runtime. I tried both local VM & VPS at vultr. both not working. 
I run on docker environment latest edition.
How to run v1.1.3 on docker ?

unocelli, 2022-01-07:
with which version are you having problems?

thingstanje, 2022-01-07:
v1.1.4 run docker edition.

unocelli, 2022-01-07:
can you try the npm version 1.1.4 ?

thingstanje, 2022-01-07:
Okey. I’ll try this morning. Do you have a step on how to make it as docker
container?

unocelli, 2022-01-07:
No, I'm sorry.
new in docker version 1.1.4 is that I build a image for more OS/ARCH and npm does it.

MikeEitel, 2022-01-15:
In node-red dashboard clearing browser cache helps..

unocelli, 2022-05-13:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2022-05-13 by unocelli