frangoteam/FUXA #251: hi, how to stop animate

echolove38, 2022-01-12:
About the animation type element
I set an action with runs when ranges 20~80,  
firstly I give a value 60, it start animate,
after a while , I give a value 1000, the animate still exist, 
so how to stop the animate
[image]

unocelli, 2022-01-12:
Hi,
you have to configure the 'stop' action, for example in the range 80~2000

echolove38, 2022-01-13:
> 
I don't think this is a good thinking, It should stop when out of the given range

unocelli, 2022-01-16:
you have to see like a trigger that start a job, the stop is more like a reset...it could be interesting when in the future there will be more animations and you want to combine them...

unocelli, 2022-10-09:
This works in the latest version. let me know if you have any issues.

closed 2022-10-09 by unocelli