frangoteam/FUXA #253: plase add  LS PLC TCP/IP Lib

AbdalaMask, 2022-01-12:
link libry 
About
LS PLC TCP/IP Lib

https://github.com/poohhw/XGTLib_LSPLC

unocelli, 2022-01-12:
Hi,
I'm sorry but I can't add the library, it should be tested and I do not know what kind of plc it is...

AbdalaMask, 2022-01-13:
[https://www.ls-electric.com/ko/upload/images/product/XGT-Series_XGK_XGI_XGR_th(1).png](url)

Rafael-OliveiraBR, 2022-09-08:
> Hi, I'm sorry but I can't add the library, it should be tested and I do not know what kind of plc it is...

@unocelli, I'm ressurecting this topic because I was about to ask you the same thing.
I wanna be able to communicate with LS PLC's over XGT protocol than modbus RTU/TCP.

I work with these devices in a daily basis, would be no problem to me to test the implementation and give feedback to you.
So frangoteam could officially add the driver in future updates.

How can we work together on this?

unocelli, 2022-10-09:
Hi, sorry for the delay,
sorry but we have to focused on the most common protocols that increase the value of the application, also I have not found usable drivers  for XGT compatible with nodejs

unocelli, 2023-05-14:
I’m closing this issue since there is nothing feasible and due to inactivity.

closed 2023-05-14 by unocelli