frangoteam/FUXA #257: BACnet - Cannot connect to the device

Skurken92, 2022-01-17:
While I find my device in YABE and try to connect to the same IP: PORT after I close YABE it still doesn't connect to the device and see the BACnet assets.

This is a Johnson Controls NAE BACnet/IP device if that has any means.

Any tips are greatly appreciated.

cicci44, 2022-02-11:
> While I find my device in YABE and try to connect to the same IP: PORT after I close YABE it still doesn't connect to the device and see the BACnet assets.
> 
> This is a Johnson Controls NAE BACnet/IP device if that has any means.
> 
> Any tips are greatly appreciated.

maybe some debug log could help us understand what the issue you are refer to? 
...din felbeskrivning hänvisar inte till något som är relaterat direkt till fuxa :)

unocelli, 2022-10-09:
Since there has been no answer to the question above, I'm assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2022-10-09 by unocelli