frangoteam/FUXA #259: Format output value

Ababuy78, 2022-01-20:
This is not an issue, it is more of an improvement.

It would be nice to be able to format the output value like a C String.Format.

For instance to show a date always with two characters like day 1, month 1and year 2022 show 01-01-2022. Or add text before and after the output value.

unocelli, 2022-01-20:
Hi,
you mean to format the datetime in chart? 
by output value you can format your value with unit (that will be added after) and digits
[image]

Ababuy78, 2022-01-20:
Hi!

Thank you for your reply! This is an amazing project, congratulations!

I mean to make a string like Format(“Program {0:00} - Segment {1:00}”,TagProgram,TagSegment) to show two tags as "Program 01 - Segment 08".

unocelli, 2022-01-20:
Yes, good idea.
I think could be nice as new output control and as alternative. I added to my todo list ;-) Thanks

unocelli, 2023-10-10:
Now the feature can be solved with a script

closed 2023-10-10 by unocelli