frangoteam/FUXA #261: import svg

patson12, 2022-01-26:
Hello,

i am going to import svg but installing in my local pc it doenst work as it works in your site.. have i miss anything?
some svg in not import properly .. just half part of svg imports

unocelli, 2022-01-26:
Hi,
can you send me the svg file please?

patson12, 2022-01-27:
plz find attchment. group and ungroup is not working after importing this file..i this in github some old code thant that in your live project .. i have downloaded code from github latest version but color change also not working in local pc.. if there is another code , can you plz upload this??

thanx .. :)
[image]

unocelli, 2022-01-27:
I don't know what you are trying to do, but currently imported svg images or grouped shapes don't support the set colors by logic, you can use the FUXA shapes, configure the logic and then group them.

patson12, 2022-01-28:
imported svg support set color in fuxa demo link.. but not working in github latest code. . can you share latest code which is in demo site..

thank you...

unocelli, 2022-01-28:
Hi,
yes I was able to reproduce the color issue, I have to investigate.
regarding the image where a piece is missing, check that you do not have any other images on front, add it in a new view.
can you tell me your os and browser used?

patson12, 2022-01-28:
it is svg. there is no any other images on front.. added in a new view. find screenshot.. 
os:- windows 10.
browser : ch
[image]
rome

unocelli, 2022-01-28:
This is from my local pc with the last master version.
[image]

The demo code is in [frangoteam/www](https://github.com/frangoteam/www)

patson12, 2022-01-28:
given demo code link is not working.. :(.. can you send me back?
and thank you so much for your help.. :)

unocelli, 2022-01-28:
Sorry,
The demo code is in [demo](https://github.com/frangoteam/frangoteam.github.io)

patson12, 2022-01-28:
my problem is solved .. thank you so much ...

patson12, 2022-01-28:
did something i miss?
after ungroup element i found thi
[image]
s

unocelli, 2022-01-28:
i suppose with the old demo library? happens on all group and ungroup images? try to clear the browser cache

patson12, 2022-01-29:
now it is working fine.. i have updated all new files..

closed 2022-01-29 by patson12