frangoteam/FUXA #264: Custom calculation or soft tags

zplorer52, 2022-01-28:
Hi, 
FUXA is a great piece of work. Are there any tutorial or template how to develop driver plugins? By the way WebApi does a lot.
How to execute a simple math formula or create some calculated soft points on existing device points values?

unocelli, 2022-01-29:
Hi, thanks for your interest,
I don't have documentation but I added a [template ](https://github.com/frangoteam/FUXA/blob/master/server/runtime/devices/template/index.js) device in the server, if you follow it, I think you will be able to understand how to create your own driver, however, please ask me for specifics by open a discussion.
Yes the soft tags will be the next task: the idea is to create scripts that can be started at specific intervals or on specific events, the scripts will be in javascript and configured in frontend with a text editor. MathJax will be supported too.
yes, svg drawn with inkscape can be imported, but I suppose that you have already tried ;-)

zplorer52, 2022-01-31:
Hi,
  Thanks for the device template. 

_"yes, svg drawn with inkscape can be imported, but I suppose that you have already tried"_ -  I have tried and failed. 

1. Exported a fresh view *fuxa-view.json* file for an unique identity. 
2. replacing "svgcontent" key value with Inkscape svg content after removing **<?xml version="1.0" encoding="UTF-8" standalone="no"?>** and also [" to \"] gives me json error.
3. I come across this [XMLSerializer](https://developer.mozilla.org/en-US/docs/Web/API/XMLSerializer) issue and understand  "svgcontent" key is expecting string value.
4. Importing view file extension (*.fuxav) is also different.
5. A little success with draw.io exported svg file but quite difficult to do post editing.

unocelli, 2022-02-04:
Sorry, I misunderstood, I thought you wanted to import images created with inkscape into fuxa.
I don't understand why you want to import views into inkscape, I've never tried it.

unocelli, 2022-05-08:
Closing this due to inactivity.

closed 2022-05-08 by unocelli