frangoteam/FUXA #277: How to start FUXA with systemd?

Daini, 2022-02-21:
Dear FUXA developers,

I installed FUXA on my Raspberry Pi and can start it with sudo fuxa.
My first project with FUXA works and connects to a Modbus-TCP device to get some registers. Excellent, the best opensource HMI I could find so far!

But I have the problem to start the fuxa script at boottime and using my project.

I tried writing a fuxa.service for systemd based on your script /usr/bin/fuxa and it starts fuxa successfully on Port 1881. But it starts with an empty project and unconfigured devices then.

Is there a way other than starting fuxa manually on a Linux system?

Kind regards

Daini, 2022-02-21:
I have a draft, not Best Practice, but it works for me at the moment.

I created a unit file for FUXA and reloaded systemd:

```
/lib/systemd/system/fuxa.service
->

[Unit]
Description=FUXA Service
After=network.target

[Service]
Type=simple
Restart=always
RestartSec=1
User=me
ExecStart=/home/me/fuxa_start.sh

[Install]
WantedBy=multi-user.target

->
#systemctl daemon-reload
```
The Executable is a bash script in my install directory, that mimics the manual start of FUXA:
```
/home/me/fuxa_start.sh
->

#!/bin/bash
cd /home/me
sudo /usr/bin/fuxa
```

To start/stop my FUXA connection to the modbus-device and the webservice on port 1881
->
```
#systemctl start/stop fluxa.service 
```
To enable/disable this service for autostart
->
```
#systemctl enable/disable fluxa.service`
```

unocelli, 2022-02-21:
Hi,
another option would be to use [pm2](https://github.com/Unitech/pm2), which also monitors the application, in this case the fuxa server #88

Daini, 2022-02-21:
Interesting, didn't know P(rocess) M(anager) 2.
Will see if I can use it. Eventually in combination with systemd.
Are you using PM2 yourself for launching FUXA server?

Thank you, amazing work!

Daini, 2022-02-22:
I could reduce the fuxa unit file by introducing the WorkingDirectory that fuxa needs as environment:

```
[Unit]
Description=FUXA Service
After=network.target

[Service]
Type=simple
Restart=always
RestartSec=1
User=me
WorkingDirectory=/home/me
ExecStart=sudo /usr/bin/fuxa

[Install]
WantedBy=multi-user.target
```

No bash script required any longer.