frangoteam/FUXA #278: Shape fill color after blink action stopped.

daisuke0604, 2022-02-22:
When I set the data in the following order with the below settings, 
I want it to be gray at the end, but it does not change from red.

test1 = 1   : Filled with gray (OK)
test1 = 100 : Filled with red (OK)
test1 = 50  : Blink start (OK)
test1 = 100 : Blink stop and filled with red (OK)
test1 = 1   : Filled with red (NG)

This only happens when blinking, not when rotating.
I'm running Version 1.1.5-625 on docker.

Settings:
[image]
[image]

unocelli, 2022-02-26:
Hi, thanks for the report, I have to fix it.

unocelli, 2022-05-08:
Let me know if something doesn't work as expected

closed 2022-05-08 by unocelli