frangoteam/FUXA #281: Limitation with Input Value control

frostpun, 2022-02-24:
"Output value" can be any data type, even String, but "Input value" only works with numerical data. Is there a way to set from input string instead of only numbers?

unocelli, 2022-02-26:
Hi,
which version you are using and what type of device and tag you have tried?

frostpun, 2022-02-28:
FUXA V.1.1.4-623
the device is an opc ua server created on my local computer, I can change the string tag remotely from my computer but when I try to use FUXA, the input value control is only accepting pushing a numerical string when I push a button to set from input.

unocelli, 2022-02-28:
You have some error message in backend or frontend?
[image]

frostpun, 2022-02-28:
no there are no errors, I am able to change the value to either numerical or string from my computer and from FUXA if the button is set, but not set from input

https://user-images.githubusercontent.com/96480690/155978614-3f3973f8-0d10-47ff-a1cd-e1d3e3ea4c1f.mp4

unocelli, 2022-02-28:
Thanks for the report. I will fix it

closed 2022-03-08 by frostpun