frangoteam/FUXA #283: Support  for infludDB

jcxjcx123, 2022-03-02:
Hi!
 I would like to know the specific steps of adding a new device connection type, so that I can add a real-time database of influxDB

unocelli, 2022-03-02:
Hi, 
Great, it would be nice to add it as a choice. 
In my opinion influxDB should replace the DAQ (Data Acquisition System) part. if you look at the server side in `./server/runtime/storage` you find index.js which is the wrapper for daqnode.js. so we should write a new daqnode that interfaces with influxDB and export functions:
- close
- setCall
- addDaqValue
- addDaqValues
- getDaqMap
- getDaqValue

I think that the structure of the db can remain the same with a table of tag properties linked to the table of values. it is also necessary to manage the influxDB service by start fuxa.
I think the first step is to manage the influxDB to start service and create the tables.

unocelli, 2022-09-08:
going to close this as resolved. let me know if you have any issues.

closed 2022-09-08 by unocelli