frangoteam/FUXA #287: how does script work?

patson12, 2022-03-04:
is there any doc about how does script code work?? 

Regards..

unocelli, 2022-03-06:
Hi,
I will add an example in the wiki as soon as possible. Anyway, I think you can e little bit intuit how it works ;)
The script must be written in javascript and for the moment you can only call it from Events of shapes and controls.

patson12, 2022-03-07:
ok :)
 
i have implemen
[image]
ted some control and shapes in older fuxa version.. so i need to implement this script code in my version. but it is in read only mode.. did i miss something?

PFA 

thanks and regards.

unocelli, 2022-03-08:
In [wiki](https://github.com/frangoteam/FUXA/wiki/HowTo-configure-Script) I added a simple example