frangoteam/FUXA #288: WebApi Result of request..

stovesy, 2022-03-04:
FUXA V1.1.6-642
Using WebAPI the result of request does not fill with the json data..

Data returned from the server is
{"time":1646389890,"number":10}
Result of request does not complete, and result spinner continues.

Data is correct in _readRequest function in httprequest/index.js
`    var _readRequest = function () {
        return new Promise(function (resolve, reject) {
            if (data.property.method === 'GET') {
                axios.get(data.property.address).then(res => {
                    lastTimestampRequest = new Date().getTime();
					console.log("HTTP Data is ", parseData(res.data, data.property));
                    resolve(parseData(res.data, data.property));
                }).catch(err => {
                    reject(err);
                });
            } else {
                reject();
            }
        });
    }
`

Thank you.

unocelli, 2022-03-06:
Hi,
I'm sorry but I don't understand what is not working...

unocelli, 2023-02-05:
I am closing this issue due to inactivity.

closed 2023-02-05 by unocelli