frangoteam/FUXA #289: value is changed continously  SiemensS7 Divice

patsonkaushik, 2022-03-04:
value is changed continously in same ip address..
plz have a look attached video..

https://user-images.githubusercontent.com/77775964/156760151-5888ca03-fad4-4f99-b623-26a42426fc66.mp4

unocelli, 2022-03-05:
Hi,
are you sure that is not a problem of some change that you have made? I tested again and for me is working :\

patsonkaushik, 2022-03-07:
value is changed continously in same address tags all tags value same type different

https://user-images.githubusercontent.com/77775964/156982439-431b24b3-6d77-42e1-9cdf-0dbfbeb45af3.mp4

e

unocelli, 2022-03-10:
is a little confusing...in the last comment the value aren't change continuously, the tags are mapped with the address...please can you me explain in which case you need to show the same value in different types?

patsonkaushik, 2022-03-11:
Please check this video   same address but same value display

https://user-images.githubusercontent.com/77775964/157856865-d890eb09-3b06-4d24-a061-52edeecfa24c.mp4

unocelli, 2022-03-11:
Sorry, but I can't help you, your description is too generic and I don't know where I have to look. it is possible that the problem is in your simulator or in the communication library node-snap7. I suggest you to try with a real siemens plc or try another application that uses the node-snap7 library like node-red so that we can restrict the search.

unocelli, 2022-10-09:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2022-10-09 by unocelli