frangoteam/FUXA #290: its possibile separate client and server side?

mircotamburini, 2022-03-04:
its possibile separate client and server side.... i would like to do an .Net Server side

unocelli, 2022-03-05:
Hi,
of course and interesting idea (#43),  in the simple diagram you see the communication with the server goes through webapi and socket.io.

[image]

mircotamburini, 2022-03-06:
Thank you, there's documentation that describe  api call method?
It can be very usefully

mircotamburini, 2022-03-06:
I start to write a Asp.Net WebApi controller , i need RequestModel / ResponseModel of every web call...

using Microsoft.AspNetCore.Mvc;

namespace Fuxa.Server
{
    [ApiController]
    [Route("/api")]
    public class ApiController : ControllerBase
    {

        private readonly ILogger<ApiController> logger;
        

        public ApiController(ILogger<ApiController> logger)
        {
            this.logger = logger;

        }

        [HttpGet("settings")]
        public async Task GetSettings(string req)
        {
           logger.LogInformation(req);
        }

        [HttpPost("settings")]
        public async Task PostSettings(string req)
        {
            logger.LogInformation(req);
        }

        /**
        * GET current Alarms 
        * Take from alarms storage and reply 
        */
        [HttpGet("alarms")]
        public async Task GetAlarms(string req)
        {
            logger.LogInformation(req);
        }

        /**
        * GET current Alarms 
        * Take from alarms storage and reply 
        */
        [HttpGet("alarmsHistory")]
        public async Task GetAlarmsHistory(string req)
        {
            logger.LogInformation(req);
        }

        /**
         * POST Alarm ACK
         * Set alarm ack
         */
        [HttpPost("alarmack")]
        public async Task PostAlarmAck(string req)
        {
            logger.LogInformation(req);
        }

        /**
         * POST Alarms clear
         */
        [HttpPost("alarmsClear")]
        public async Task PostAlarmClear(string req)
        {
            logger.LogInformation(req);
        }

        /**
         * POST SignIn
         * Sign In with User credential
         */
        [HttpPost("signin")]
        public async Task PostSignin(string req)
        {
            logger.LogInformation(req);
        }

        /**
         * GET daq data
         * Take from daq storage data and reply 
         */
        [HttpPost("daq")]
        public async Task PostDaq(string req)
        {
            logger.LogInformation(req);
        }

        /**
         * GET Server logs folder content
         */
        [HttpGet("logsdir")]
        public async Task GetLogsDir(string req)
        {
            logger.LogInformation(req);
        }

        /**
         * GET supported Plugin and status (installed) 
         */
        [HttpGet("plugins")]
        public async Task GetPlugins(string req)
        {
            logger.LogInformation(req);
        }

        /**
        * POST Plugin
        * Install the plugin
        */
        [HttpPost("plugins")]
        public async Task PostPlugins(string req)
        {
            logger.LogInformation(req);
        }

        /**
        * DELETE Plugin
        * Unistall the plugin
        */
        [HttpDelete("plugins")]
        public async Task DeletePlugins(string req)
        {
            logger.LogInformation(req);
        }

        /**
         * GET Project data
         * Take from project storage and reply 
         */
        [HttpGet("project")]
        public async Task GetProject(string req)
        {
            logger.LogInformation(req);
        }

        /**
        * POST Project data
        * Set to project storage
        */
        [HttpPost("project")]
        public async Task PostProject(string req)
        {
            logger.LogInformation(req);
        }

        /**
        * POST Single Project data
        * Set the value (general/view/device/...) to project storage
        */
        [HttpPost("projectData")]
        public async Task PostProjectData(string req)
        {
            logger.LogInformation(req);
        }

        /**
        * GET Project demo data
        * Take the project demo file from server folder 
        */
        [HttpGet("projectdemo")]
        public async Task GetProjectDemo(string req)
        {
            logger.LogInformation(req);
        }

        /**
        * GET Device property like security
        * Take from project storage and reply 
        */
        [HttpGet("device")]
        public async Task GetDevice(string req)
        {
            logger.LogInformation(req);
        }

        /**
        * POST Device property
        * Set to project storage
        */
        [HttpPost("device")]
        public async Task PostDevice(string req)
        {
            logger.LogInformation(req);
        }

        /**
        * POST Upload file resource
        * images will be in media file saved
        */
        [HttpPost("upload")]
        public async Task PostUpload(string req)
        {
            logger.LogInformation(req);
        }

        /**
         * POST runscript
         * Run script, can be call with script id or script content as test 
         */
        [HttpPost("runscript")]
        public async Task PostRunScript(string req)
        {
            logger.LogInformation(req);
        }

        /**
         * GET Users
         * Take from users storage and reply 
         */
        [HttpGet("users")]
        public async Task GetUsers(string req)
        {
            logger.LogInformation(req);
        }

        /**
         * POST Users
         * Set to users storage
         */
        [HttpPost("users")]
        public async Task PostUsers(string req)
        {
            logger.LogInformation(req);
        }

        /**
         * DELETE User
         * Set to project storage
         */
        [HttpDelete("users")]
        public async Task DeleteUsers(string req)
        {
            logger.LogInformation(req);
        }
    }
}

unocelli, 2022-03-06:
I hope it can help you. please let me know if you find any errors ;)
[FUXA_WebAPI.pdf](https://github.com/frangoteam/FUXA/files/8192634/FUXA_WebAPI.pdf)

mircotamburini, 2022-03-06:
yeah it's exactly what i need :)

mircotamburini, 2022-03-30:
which port use socket.io fuxa client

unocelli, 2022-03-30:
I don't set a port: `const io = socketIO(server);` so it should be the same as the web server (1881).

closed 2022-10-09 by unocelli

luy710, 2023-11-27:
> I start to write a Asp.Net WebApi controller , i need RequestModel / ResponseModel of every web call...
>
> using Microsoft.AspNetCore.Mvc;
>
> namespace Fuxa.Server { [ApiController] [Route("/api")] [...]

Where is the .NET Core version? Can I participate in it?