frangoteam/FUXA #292: OPC UA polling time is not 200ms, but around a second

AgrisNN, 2022-03-09:
Dear FUXA team,

FUXA V.1.1.5-625

There's slow behavior of OPC UA FUXA plugin connected to S7-1200 PLC.
FUXA device setting is set to poll at 200ms, but actual values are updated too slow, seems around 1s.
I'm sure PLC reports value updates as fast as it can around 100ms.
Another OPC UA Client (namely, UAExpert) displays value updates as it has to be.

On this video you may see "byte" variable which changes own values, but "bool" vars are stuck, even OPC server changes them false\true at rate 1Hz for "myClock1Hz" and 2Hz for "Clock_2Hz", respectively. 

[image]

https://user-images.githubusercontent.com/101255573/157435930-8efe0422-3a72-4fda-845d-d78868b1b256.mp4

Any suggestions?

unocelli, 2022-03-09:
Hi, Thanks for the report.
you are right the polling settings is the interval to read the values from the device and send them to the frontend, but for opcua device I use the monitor subscription (callback of changed value) and this is fix in code with 1 second sampling interval :( I have to improve it.
` 
var monitoredItem = await the_subscription.monitor(
   { nodeId: nodeId, attributeId: opcua.AttributeIds.Value },
   { samplingInterval: 1000, discardOldest: true, queueSize: 1 },
   opcua.TimestampsToReturn.Both
);
`

AgrisNN, 2022-03-10:
Great news! Thanks.

unocelli, 2022-05-08:
Let me know if something doesn't work as expected

closed 2022-05-08 by unocelli

haitan981, 2022-10-07:
I'm getting the same error as you, but in the index.js file, I can't find the part that needs fixing like yours. Can you take a screenshot of the code you edited to fix that error?
Thank you so much
[image]

> Great news! Thanks. I confirm that manual editing server/runtime/devices/opcua/index.js and changing samplingInterval from 1000 to 200 resolves the problem.

AgrisNN, 2022-10-08:
> I'm getting the same error as you, but in the index.js file, I can't find the part that needs fixing like yours. Can you take a screenshot of the code you edited to fix that error? Thank you so much [image]

You'll find it here, red arrow points:
[image]
 
Probably you may change it in settings somehow (data.polling variable), but I have no idea where they're stored.
May be [unocelli](https://github.com/unocelli) knows...

haitan981, 2022-10-10:
> > Tôi cũng gặp phải lỗi giống như bạn, nhưng trong tệp index.js, tôi không thể tìm thấy phần cần sửa như lỗi của bạn. Bạn có thể chụp ảnh màn hình đoạn mã bạn đã chỉnh sửa để sửa lỗi đó không? Cảm ơn bạn rất nhiều !
> 
> Bạn sẽ tìm thấy nó ở đây, mũi tên màu đỏ trỏ: [image]
> 
> Có thể bạn có thể thay đổi nó trong cài đặt bằng cách nào đó (biến data.polling), nhưng tôi không biết chúng được lưu trữ ở đâu. Có thể [không](https://github.com/unocelli) biết ...

I fixed it to 100 but didn't see a change in the sampling rate

malcore10, 2022-10-10:
The sampling rate depend from your server too, you have to look if it support 100 msec.
The best opcua server have difficulty to go under 200 msec.