frangoteam/FUXA #297: svgeditor  opensource

VansirCQ, 2022-03-14:
The files under svgeditor are basically processed in binary. Is it necessary to add SVG through this editor component? Can the source code of this part be open source? I want to add a custom SVG component to FUXA. Do you have any questions? any suggestion, thanks a lot

VansirCQ, 2022-03-15:
> 
emmmm..... " example how I added the iframe control ", where is it?  
When I see an issue in 2019, you want to open source this fuxaeditor. It seems that fuxa fans have been waiting too long. Hahaha, when can you open source it? Let's improve it together

unocelli, 2022-03-18:
I accept the criticism...I wonder how you plan to help me if you haven't understood how to add your own control just by looking in the code :wink:

closed 2022-03-18 by unocelli

xiaote0225, 2023-02-14:
@unocelli Hi unocelli,I also have the same questions?1.How to add your own control in FUXA? Is there any document or video？2.When do you want to or  plan to publish fuxa-editor? 

Looking forward to your reply.

unocelli, 2023-02-14:
fuxa-editor is [svg-edit](https://github.com/SVG-Edit/svgedit) packaged to work with angular.

concept10, 2023-02-14:
> @unocelli Hi unocelli,I also have the same questions?1.How to add your own control in FUXA? Is there any document or video？2.When do you want to or  plan to publish fuxa-editor? 
> 
> 
> 
> Looking forward to your reply.
It took me a couple of weeks of digging in the code, but building components are not difficult, especially with all of the Angular examples out there. 

It's really not that difficult to follow and duplicate what's already there. The SVGEditor source alone will not get you anywhere.

xiaote0225, 2023-02-15:
Hi Tracey,thanks for you answer.
Our urgent requirement is to extend the echarts chart into fuxa.
You mean that I can integrate echarts by looking at the existing code in the Fuxa project, right?

发自我的iPhone

concept10, 2023-02-15:
@xiaote0225 I was looking to accomplish the same a few weeks back but took another route. I'm going to give it another shot.

unocelli, 2023-02-15:
An example: the Bar graph control was added without modifying the editor library. [image]

xiaote0225, 2023-02-15:
Thank you very much.It means that I can insert a echarts control in FUXA follow the bar graph control without modify the editor library，right？

发自我的iPhone

xiaote0225, 2023-02-16:
> 

@concept10 , Could you specify your idea. Maybe we can cooperate to achieve this goal.

unocelli, 2023-02-16:
Yes, you can

xiaote0225, 2023-02-23:
@unocelli Hi, guy,I am viewing code related to socket.io. I feel that the logic of this block of data processing code is complex. Do you have any good suggestions.

unocelli, 2023-02-28:
Hi, to add your own control, you must not use or modify anything related to socket.io