frangoteam/FUXA #298: How is fuxa deployed in the background of demo link

VansirCQ, 2022-03-14:
How do you realize that every time you click on this website( https://frangoteam.github.io/  ), it will be an initialized demo page. I want to make a service using isolation. Different users click in the initialized page. They are not affected by each other. A lot of services are started in the background with docker? Thank you for answering my question

unocelli, 2022-03-14:
Hi,
the demo page is only frontend hosted by github as static website, don't have a backend, it is only to show the editor functionality, by the way I should update it.

VansirCQ, 2022-03-15:
> Hi, the demo page is only frontend hosted by github as static website, don't have a backend, it is only to show the editor functionality, by the way I should update it.

alright, I get it, If I want to make fuxa service-oriented and user operations do not affect each other, then I may need to use the idea of docker to enable several services in the background