frangoteam/FUXA #300: Internal Plugin is missing

eyadalhanafi, 2022-03-16:
Thank you for your support.
 
When i created Internal device plugin, I got the below warning message 
[WAR]  try to create LocalTags but plugin is missing!

Fuxa version: 1.1.5-642

unocelli, 2022-03-16:
Hi,
I don't understand, you don't create internal device plugin, you install the plugin? which one?

eyadalhanafi, 2022-03-17:
In connections, i created a device type "internal". I defined device tags.
when run the application i started getting a warning "Plugin is missing"
hope it is clear now.

unocelli, 2022-03-17:
Hi,
Yes, I can reproduce it, the warning message is wrong because the internal device is only in frontend (don't need plugin). I will fix it...Thanks

unocelli, 2022-05-08:
Let me know if something doesn't work as expected

closed 2022-05-08 by unocelli