frangoteam/FUXA #302: smartphone

JCVD1, 2022-03-18:
dear
I don't know if there is already a question about it but is it possible to open the fuxa server on another device e.g. your cell phone?

unocelli, 2022-03-18:
hi,
I'm not sure I understand the question. Can you give me an example?

JCVD1, 2022-03-21:
is it possible to open your visualization of fuxa on a device other than your local computer. For example on your smartphone or a computer other than local.
If so, do you have to run it on a different server than your local from your computer or do you have to install something extra on your smartphone?

unocelli, 2022-03-21:
yes, it is in standard html5 and you can view it with the browser of modern smartphone, try your self [https://frangoteam.github.io/](https://frangoteam.github.io/) . I don't recommend to use the editor.

studuino, 2022-03-26:
@JCVD1 
Hi, 
Your question is interesting.
there is basically two parts in Fuxa
(1) Frontend(client)
For frontend user interface, you can use almost any standard browsers.

(2) Backend(server)
For backend server, you may try with linux emulator Termux (android) or ish(ios).

I've never tried myself but I am interested.

unocelli, 2022-10-09:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2022-10-09 by unocelli