frangoteam/FUXA #304: How to switch between two multi views containers and how to open a card on a position specified by user?

IonuDragomir1, 2022-03-21:
Hello!

Could you please help solving these two issues?
1. I can't switch between two multi views containers. For example, I have two  multi views containers (MVC1 and MVC2) and a Canvas/SVG view. I can't switch directly from MVC 1 to MVC2. To do so I must switch from MVC1 to Canvas/SVG view and then to MVC2.
2. If I open a card when pressing a button, it opens on a random position. How could I open the card on a position specified by user?
Many thanks in advance!

unocelli, 2022-03-21:
Hi, thanks for the report.
1. is a bag that I have to fix.
2. and 3.  are new features, they are not implemented.
I would appreciate if you could separate the 3 issues so that they are clearer to manage.

unocelli, 2022-03-22:
#306 #307 #308

closed 2022-03-22 by unocelli