frangoteam/FUXA #305: Chart control problem

lifeihub, 2022-03-22:
Hello, can you optimize the chart control? When the control is zooming in and out, there will be adaptive problems. The font has no adaptive size, and the most important thing is that the curve will be lost in the control. I hope you can optimize it when you have time. Thank you very much and look forward to your reply.
[image]

lifeihub, 2022-03-22:
Hello, I hope you have time to optimize this control
[image]

unocelli, 2022-03-27:
Hi, sorry for the delay.
what do you propose, I can't change the font, maybe we can limit the zoom?

lifeihub, 2022-03-28:
> 你好，抱歉耽搁了。 你有什么建议，我不能改变字体，也许我们可以限制缩放？

Well, first of all, thank you for your reply. I think the way to change is to introduce three-party chart controls. At present, this adaptive situation can not meet the demand. If you have a better method, I look forward to your reply. Thank you again.

unocelli, 2022-03-30:
I think uplot is light, simple and performant, a good compromise for an application like fuxa. 
I am open to other suggestions but I don't have resource.

closed 2023-05-14 by unocelli