frangoteam/FUXA #308: Found issues when using same view for repeating components.

IonuDragomir1, 2022-03-22:
I have seen in [your example](https://github.com/frangoteam/FUXA/wiki/HowTo-use-same-view) that I can use the same view on different objects. Your example is working fine because you close Valve A card and then open Valve B card. If you'd open Valve B card while Valve A card is opened, the text "Settings Valve A" is replaced with "Settings Valve B".  I didn't test if the tags (Modbus-A-1 and Modbus-B-2) associated with input field for Open % (temp value) are mixed up.

IonuDragomir1, 2023-02-20:
Hi, @unocelli!
Any updates on this topic? It's a pretty much important bug. I have an application where 200 same views with different objects must be opened.

unocelli, 2023-02-25:
Hi,
what is your purpose, to keep the same card open several times?

IonuDragomir1, 2023-04-04:
Hi,
Sorry for the delay. It's now when I've seen your question.
The problem is not about opening the same card several time (although it would be useful if the card had tabs and  the user could see in parallel same card's tabs). My problem is the same as @robsori described in [#744](https://github.com/frangoteam/FUXA/issues/744).